leaphy-robotics/leaphy-blocks #10: Standaard google blocks uit blocks repo verwijderen

rmoesbergen, 2023-10-28:
We hebben nu kopieen gemaakt van de standaard google blocks (color, lists, logic, loops, procedures, etc.). Deze gaan uit de pas lopen wat voor bugs zorgt. Laten we ze verwijderen de standaard blocks uit Blockly gebruiken (de code/def is 1 op 1 als het goed is).

JeftadeGraaf, 2023-12-05:
Ik heb inmiddels de code aangepast dus nu worden de definitions van blockly gebruikt.

Problemen:
- De styles van thinkflow en numbers is nu zwart, heb even gekeken of het ergens nog kan worden aangepast maar zag niet echt een goede oplossing. https://groups.google.com/g/blockly/c/byDoKnftEcc
- De definition van controls_repeat_forever staat niet in google blockly, heb hem nu toegevoegd in arduino.js. Weet het niet zeker maar dacht dat de generator ervan al in main stond.

rmoesbergen, 2023-12-15:
Ik heb je branch uitgechecked en uitgeprobeerd in webbased. Ik denk dat er iets te veel weg is nu, want ik krijg een error dat de generator van een block niet gevonden kan worden (kom er nog niet helemaal achter welke dan). Dezelfde error die jij kreeg en in de whatsapp groep gooide. Ik zou verwachten dat de definities wel standaard zijn (dus de json/js van de blocks), maar dat generator code wel moet blijven. Er is immers geen standaard C++ generator in Blockly. Waarschijnlijk zijn er wel een aantal blocks die we (nog) niet gebruiken, waar wel een generator voor is.

JeftadeGraaf, 2024-01-02:
Oke, ik ben er na een hele lange tijd achter gekomen dat het toch wel vrij makkelijk op te lossen was. Heb al een PR aangemaakt. De blokken staan nu nog wel inde folder maar worden niet ingeladen

closed 2024-03-27 by sverben