leaphy-robotics/leaphy-blocks #131: DCmotor frequency

boutreet, 2024-10-27:


rmoesbergen, 2024-12-08:
Did some research. We use pin 3 and 11 for the nano DC motor PWM pins. These are both attached to Timer2 on the AVR328P. So far so good, however:

```
Output pins for Timer2 are PORTB pin 3 and PORTD pin 3. On Arduino, these are digital pins 11 and 3. But the library supports PWM only for pin 3 (PWM_PIN_3). Because in Timer Mode 5 (PWM, Phase Correct) the register OCRA is used to save the top value of the timer. The duty cycle is specified as a 8 bit value, so anything between 0 and 255.
```

So pin 11 cannot use PWM mode with this timer :(

rmoesbergen, 2024-12-24:
Created pr #145 to work around this by mapping the speeds to higher PWM values for nano based robots

closed 2024-12-24 by rmoesbergen